ros-industrial-consortium/descartes #46: descartes robot model as plugin

jrgnicho, 2015-01-13:
In order to use RobotModel implementations in a generalized way then each Implementation should be made into a plugin that can be loaded at run time.  Currently this won't be possible due to a limitation that prevents libraries containing the plugins from being loaded dynamically if they are also linked statically.  The descartes_core package where the RobotModel interface resides suffers from this drawback since it also contains the implementation for the various trajectory point classes which are linked statically.  One easy solution would be to isolate the RobotModel class into its own package.  The [nav_core] (https://github.com/DLu/navigation/tree/hydro/nav_core) of the navigation repo is a good example of this.

jrgnicho, 2015-01-19:
I propose to divide the descartes packages in the following manner:
* descartes_core: Only interfaces such as:
a. TrajectoryPt
b. RobotModel
c. PlannerInterfaces (not yet defined)
d. Any future interfaces that may be needed.

*. descates_trajectory: Existing TrajectoryPt implementations. These include
a. CartesiantrajectoryPt
b. JointTrajectoryPt 
c. AxialSymetricPt (not in descartes)
d. Also any future implementations from this interface

* descartes_planner:  This will require the definition of a new interface for all planners but it should include the these two currently available planners:
1. PlanningGraph
2. SparsePlanner

* descartes_robot
1. MoveitPlannerAdapter
2. Robot specific implementations with their unique close form ik code

This change will most likely break packages in descartes and the amos repo but in the current state of things this can be easily addressed by replacing namespaces.

Jmeyer1292, 2017-04-06:
This was implemented a long time ago. See #49.

closed 2017-04-06 by Jmeyer1292