ros-industrial-consortium/descartes #48: Improvement to handle large joint changes

jrgnicho, 2015-01-23:
Paths returned by the graph planner sometimes contain large joint changes between consecutive points

shaun-edwards, 2015-01-23:
The planners should check if the joint changes are possible.  That is whether the robot is physically capable of making such a jump.  Since we are planning for trajectory points (i.e. points with timing data), this should be possible.  Unfortunately, it is more complex than it appears.  Whether a point is physically reachable depends upon the joint velocities at the current point (which is not known to the planner until after the path is planned).  

A possible compromise solution is to eliminate large changes that would require greater than the max velocities.  This essentially ignores acceleration constrains.

A proper solution would require sampling waypoint trajectories as part of the graph, in addition to current tolerances.

Jmeyer1292, 2015-02-23:
I have found some limited success in changing the way sampling works. Instead of calling robot_state's getRandomPositions() function, I call getRandomPositions(jointgroup*, RandomNumberGenerator).

Right now I'm creating a new random generator for every call to getAllIK and seeding it with a fixed value. It's not efficient or gauranteed to work, but I find it produces much better paths for what I'm working with. 

http://docs.ros.org/indigo/api/random_numbers/html/classrandom__numbers_1_1RandomNumberGenerator.html

Jmeyer1292, 2017-04-06:
The timing constraints nominally take care of this. Now practically there are other concerns. The other, more subtle, issues should be discussed in a separate PR.

closed 2017-04-06 by Jmeyer1292