ros-industrial-consortium/descartes #51: Interpolator planner

jrgnicho, 2015-02-06:
The current descartes planner (PlanningGraph, SparsePlanner) only operate on dense trajectories.  Thus an InterpolatorPlanner would produce intermediate points between the sparse waypoints and then generate a plan for the interpolated trajectory.  In addition, it should allow for several interpolation strategies (cartesian, joint, etc) and tolerances (waypoint zone) for each waypoint.

shaun-edwards, 2015-02-09:
@jrgnicho, we need more detail on this one.  I'd like to point a few people to this bug, but there needs to be more details.  What's the input?  What's the output?  What data structures are envisioned?  Can we point to existing robot languages as an example?

Jmeyer1292, 2017-04-06:
While useful, I feel that this sort of functionality belongs to a higher level package that wraps descartes as it is now.

closed 2017-04-06 by Jmeyer1292