ros-industrial-consortium/descartes #52: Moveit GetCartesianPath Service

jrgnicho, 2015-02-06:
Create a node that services the [moveit_msgs::GetCartesianPath] (http://docs.ros.org/api/moveit_msgs/html/srv/GetCartesianPath.html) and uses a descartes path planner

madratman, 2015-03-26:
I'm new here. Here's an attempt.  
https://github.com/madratman/descartes/blob/issue_%2352_moveit_cartesian_path_service/descartes_moveit/src/cartesian_path_service_trial.cpp
Sorry, I didn't see there was a separate branch for the issue already before working on it and hence I haven't used a class. And came to know later that there is a  descartes/descartes_trajectory/src/cartesian_interpolator.cpp as well.  

I have translated from moveit to descartes, inspired by the moveitstateadapter, i.e.:
            `std::string group_name = req.group_name;
	    robot_state::RobotState start_state;
	    robot_state::robotStateMsgToRobotState(req.start_state, start_state);
	    moveit::core::RobotModelConstPtr robot_model = robot_state->getRobotModel();
	 	const moveit::core::JointModelGroup* joint_model_group_ptr = robot_state->getJointModelGroup(group_name);
	 	const std::vector<std::string>& link_names = joint_model_group_ptr->getLinkModelNames();
   		tool_frame = link_names.back();
   		world_frame = robot_state->getRobotModel()->getModelFrame();`

For the GetCartesianPath's req.waypoints, I've translated them into a vector of shared_ptr to descartes_core::TrajectoryPt, inspired from godel/pathplannig. 

Further, How can the remaining request part of teh GetCartesianPath srv be utilized?  
 `float64 max_step   
 float64 jump_threshold   
 bool avoid_collisions           
 moveit_msgs/Constraints path_constraints`

jrgnicho, 2015-03-26:
@madratman  Thank you for your contribution, I'll take a look at it soon.

madratman, 2015-03-26:
Thanks. Although, I am afraid it isn't useful as of now. One more thing - talking about interpolators, there are many possible ways  -  linear, parabolic and cubic blends, bezier curves, right? Are there any plans to implement them? 
Existing/deprecated repos:
[industrial_trajectory_filters](http://wiki.ros.org/action/fullsearch/industrial_trajectory_filters?action=fullsearch&context=180&value=linkto%3A%22industrial_trajectory_filters%22)
ros : trajectory_filters, which seems quite useful [1](http://wiki.ros.org/trajectory_filters) [2](http://docs.ros.org/groovy/api/constraint_aware_spline_smoother/html/) 

Then, there are time parameterization issues in MoveIt! that Descartes could improve upon:
[Path time parameterization yields non-smooth time evolution #167](https://github.com/ros-planning/moveit_core/issues/167) 
[Planning direct path plans - thread on googlegroup](https://groups.google.com/forum/#!msg/moveit-users/_C8ysk6vURg/jL4mrVITU14J)
[Feature request: Add better functionality for adding Time Parametrization to a Cartesian Trajectory #141](https://github.com/ros-planning/moveit_core/issues/141)
[IDEXX-SwRI Path Planning Optimization ROS-I Consortium FTP](http://static1.squarespace.com/static/51df34b1e4b08840dcfd2841/t/53c3435ce4b00a24507b341d/1405305692771/IDEXX_Planning_Presentation.pdf)

jrgnicho, 2015-03-26:
I'm currently working on a branch that implements interpolation in cartesian space with blends regions.  Since this is an initial implementation it will only provide limited interpolation options so we'll definitely be looking to incorporate other interpolation options  in the future.

Jmeyer1292, 2017-04-06:
This integration may eventually be part of a larger integration that depends on this package.

closed 2017-04-06 by Jmeyer1292