ros-industrial-consortium/descartes #66: all: pkgs using C++11 features should export their cflags (and check compiler support)

gavanderhoorn, 2015-03-09:
All pkgs in the `descartes` repository add the `-std=gnu++0x` compiler option somewhere in their `CMakeLists.txt` ([descartes_core](https://github.com/ros-industrial-consortium/descartes/blob/672b9c0907755ef23bfc92905942425cc31e19e0/descartes_core/CMakeLists.txt#L5) fi). Dependant pkgs also need to set this flag, in order to be able to use headers from `descartes` ([godel_path_planning](https://github.com/ros-industrial-consortium/godel/blob/eea4ad584052ed29442d6b29bea1a8c0c27430e7/godel_path_planning/CMakeLists.txt#L20) fi).

It is customary in such cases (in `catkin` projects) for the providing pkg to export any necessary `cflags` by exporting a `pkgname_CFLAGS` or `pkgname_DEFINITIONS` variable (with a preference for the latter, it seems) in a `.cmake` snippet that is passed to dependants using the `CFG_EXTRAS` argument to the `catkin_package(..)` call.

All pkgs in `descartes` should use this mechanism to export the required `-std=gnu++0x`.

---

Related: afaict, the `-std=gnu++0x` option could actually be GCC specific and should ideally be guarded by a check for `CMAKE_COMPILER_IS_GNUCXX` (although apparently CLang is compatible with GCC command line options, so it should work there).

gavanderhoorn, 2015-03-09:
Alternatively, the `.cmake` snippet could append the option directly to the `CMAKE_CXX_FLAGS` variable itself, which would avoid the situation where dependants forget to add an `add_definitions(${descartes_core_DEFINITIONS})` line to their `CMakeLists.txt`.

I don't like that too much though, as simply `find_package(..)`-ing the pkgs would then lead to the flags being updated automatically, which is probably not what the user wanted. Additionally, there are several buildtype-specific variants of `CMAKE_CXX_FLAGS`, which would require some if-else work to figure out which one to update from the snippet.

jrgnicho, 2015-03-09:
I'm not too familiar with the use of .cmake snippets for exporting package specific flags, could these flags (- std=gnn++0x in this case)  not be defined and exported from the CMakeList.txt file in the package?

gavanderhoorn, 2015-03-09:
I'll submit a PR later.

Re: setting them from the `CMakeLists.txt`: no, since that file is not imported by the dependants. The `CFG_EXTRAS` snippet is automatically imported.

commit ea3864d9 (2015-03-21): Remove superfluous '-std=gnu++0x' add_def() from build scripts. Fix #66.

This is now handled automatically be descartes_core.
closed 2015-03-21 by shaun-edwards