ros-industrial-consortium/descartes #77: Slow Data Structures in IK Calculations

Jmeyer1292, 2015-03-22:
While playing with Descartes to do Cartesian motion through free space, I was generating some cartesian points with several hundred to tens of thousands of joint solutions (using IKFast). I noticed that the IK process was taking a while but the actual joint solutions came available very quickly.

I did some profiling and it seems that a single IK solution takes ~5 microseconds to calculate (using IKFast). Creating a JointTrajectoryPt, putting it into a std::list, and inserting it into the joint solutions std::map takes 1 millisecond per point. As you can imagine this adds up quickly.

We pay 100x more in data structure costs than the raw calculations. I need to look into it more, but I imagine that the majority of the cost comes from cache misses associated with std::list and map as well as the cost of doing map lookup and insertion for every pt. 

I'm not sure what the right solution is, but it might require significant changes to PlanningGraph. It probably means doing away with the extra map of uuids to joint solutions. If we ever find that PlanningGraph is too slow, I believe this is the best approach to fixing that.

shaun-edwards, 2015-03-22:
Good find...just a couple of questions?

- Is this a problem in general, or just in the case of free space?  
- Are the `std::list` and `std::map` insertions for bookkeeping or do those structures store the planning graph operated on by the boost solvers?  
- Can you provide a link to the code/section where you think the problem is occurring?

gavanderhoorn, 2015-03-22:
> I did some profiling and it seems that a single IK solution takes ~5 microseconds to calculate (using IKFast).

As expected.

> Creating a JointTrajectoryPt, putting it into a std::list, and inserting it into the joint solutions std::map takes 1 millisecond per point. As you can imagine this adds up quickly.

That is rather a lot. There must either be some significant cache thrashing going on, a profiling error or something else is going on.

@Jmeyer1292: what did you use for profiling? `std::chrono`, or a profiling tool?

Jmeyer1292, 2015-03-22:
@gavanderhoorn I did poor man's profiling with chrono's steady clock, summed up the cost and iterations for a given operation and displayed their average. I'll go back through and run the planning node with cachegrind and see what it says.

@shaun-edwards 
1. This is a general problem but it shows up more when you have looser constraints on your trajectory points.
2. PlanningGraph is implemented using two maps. The first one, cartesian_point_link_, maps UUIDs to the trajectory points the user inserts with extra info about the previous point, next point, and a list of all the UUIDS of joint solutions at this point. The second one is joint_solutions_map_ and it maps UUIDs to descartes' joint_trajectory_pts. The boost graph stores just UUIDs inside its vertexs and edge weights. For every user point, we get all the IKs and then for each IK we make a TrajectoryPtPtr, push its ID into a std::list to go into the cartesian_point_link_, and then insert the point itself into the joint_solutions_map_.

3. I need to learn to use GitHub better, but here:
https://github.com/ros-industrial-consortium/descartes/blob/hydro-devel/descartes_planner/src/planning_graph.cpp#L785-L792
That loop takes on the order of 1 ms per loop iteration with my limited testing.

gavanderhoorn, 2015-03-22:
If you really want to know what is going on, I suggest using something like [perf](https://perf.wiki.kernel.org/index.php/Main_Page) with a build with debug symbols (ie: `RelWithDebInfo`), or a similar tool. Naively: I see a `new` and two copies. A proper profiler should be able to precisely tell us what is going on there.

Jmeyer1292, 2015-03-22:
Using Valgrind and RelWithDebInfo, it seems that, as usual, I'm wrong. It looks like almost all of the cost is in the UUID generator. Feel free to run tests yourself.
[image]

dpsolomon, 2015-03-22:
I would assume that a non-random uuid constructor might be more efficient,
if it uses the clock or some such. Because you're not merging points from
different computers that should be just as unique. Then again, it might not
help that much if the process lock gets called for that one too...

Jmeyer1292, 2015-04-03:
Should this be closed?

shaun-edwards, 2015-04-03:
Fixed in PR #96

closed 2015-04-03 by shaun-edwards