ros-industrial-consortium/descartes #78: Shared pointer usage in Planners

Jmeyer1292, 2015-03-25:
The current interface (and implementations) for descartes path planners defines an initialize method which takes as an argument a reference to a shared pointer to a const robot model. This implies that the interface might make changes to the shared pointer itself. It also prevents users from initializing a shared pointer to a robot model and then simply passing it into the constructor (as that requires the compiler to make two levels of indirection to ConstPtr and then to ConstPtr&).

The fix is to simply take the shared pointer by value and use std::move (since we're in c++11 land now) to get it all the way down the chain to planning graph.

jrgnicho, 2015-03-25:
How about passing the robot model by reference as done [here](https://github.com/ros-industrial-consortium/descartes/blob/hydro-devel/descartes_core/include/descartes_core/trajectory_pt.h#L107)?

Jmeyer1292, 2015-03-25:
The planners take a copy of the robot model pointer. I assume the reason that shared pointers are used in the first place is so that the lifespan of the model is tied to the lifespan of the planner. This change just makes it clear that the planner participates in ownership of the robot model and also allows users to pass shared pointers to their robot model without jumping through hoops.

shaun-edwards, 2015-03-26:
Just for some history...the original implementation might have used a raw pointer.  These were all changed to shared pointers in order to remove raw pointers from the public interface.  I wouldn't read too much into ownership implications.  My suggestion is to do what is right.  I would also prefer that this change only deprecate the existing method (this seems easy enough).

Jmeyer1292, 2015-03-26:
This change should not break any existing code.

closed 2015-03-26 by Jmeyer1292