ros-industrial-consortium/descartes #88: A* Search

ClayFlannigan, 2015-03-30:
When we first considered using A* to search the graph, we assumed it would not make much impact due to the fact that at every Cartesian point, we would need to essentially expand all the sampled configuration positions (neighbors in the graph).  But, now that we have implemented a limit on the maximum joint velocity, thereby limiting the maximum difference in joint positions between points, the number of nodes to expand will be limited.  So, should we consider A* again?

Jmeyer1292, 2015-03-30:
I'm not sure, but I do not believe that A* will help us much here. The primary reason is the graph we build is very linear. A given cartesian pose is expanded into a bunch of IK solutions (our vertices) and the 'neighbors' of each of those solutions is a set of solutions for the next cartesian pose. We could choose not to create UUID's for IK's that don't have an edge, but users are allowed to modify the trajectory in place and those points could become relevant again.

ClayFlannigan, 2015-03-30:
I guess I am proposing that you wouldn't expand (generate IK, UUID) each sampled Cartesian pose unless it is visited by A*.  Instead, you would only expand the next "nearest" Cartesian poses, where nearest might be defined by the closest pose in x, y, z, i, j, k (at most 12 nodes).

In a 2D Manhattan world, when you expand a new node, there are at most 4 neighbors (+x, -x, +y, -y), but one has already been visited, so really it is at most 3 neighbors.  In our 6D case, I think it would be the same (11 nodes maximum).  If the world was not Manhattan, then there are more, but it is bounded.  Maybe this is would only be helpful if you had a very dense sampling?