ros-industrial-consortium/descartes #95: Path planner tests should be refactored

shaun-edwards, 2015-03-31:
The path planner tests need to be improved/refactored:
 - The [dense_planner](https://github.com/ros-industrial-consortium/descartes/blob/hydro-devel/descartes_planner/test/dense_planner.cpp) & [sparse_planner](https://github.com/ros-industrial-consortium/descartes/blob/hydro-devel/descartes_planner/test/sparse_planner.cpp) tests have similar code (i.e. multiple definitions of `ThreeDOFRobot`) and similar tests.
    - The `ThreeDOFRobot` class should be extracted into it's own source/header file
    - A typed test approach, similar to [robot_model_test](https://github.com/ros-industrial-consortium/descartes/blob/hydro-devel/descartes_trajectory/test/descartes_trajectory_test/robot_model_test.hpp) should be investigated.  See [type parameterized tests](https://code.google.com/p/googletest/wiki/AdvancedGuide#Type-Parameterized_Tests) for gtest documentation.

- Tests should be added to track planner performance and make certain that a major regression does not take place.
- Tests that exercise unsolvable paths (i.e. exceeding joint velocity limits) should also be added.
- Stress tests (i.e. large number of points, unconstrained paths, etc..) should also be added (no test should take more than 10 minutes to complete).

shaun-edwards, 2015-03-31:
@madratman, this one is a little complex, but if you are willing you could address this one in pieces (several PRs).

madratman, 2015-04-01:
Thanks for this. I'll get my hands on it this week.

shaun-edwards, 2015-04-05:
@madratman, did you have a chance to look at this one?  If you could get a PR in early this week, that would be great.  It doesn't have to be a full fix.  Addressing one or part of the issue would go a long way.  Let me know one way or the other.  Also, let me know if you need any help/direction.

Jmeyer1292, 2017-04-06:
This issue is at least partially addressed by #135. I will close and open new issues more specific to tracking planner performance and stress testing.

closed 2017-04-06 by Jmeyer1292