ros-industrial-consortium/descartes #99: Boost atomic not present in ROS hydro

shaun-edwards, 2015-04-05:
The fix in PR #96 introduced a dependency on the `boost::atomic`, see [here](http://www.boost.org/doc/libs/1_57_0/doc/html/atomic.html).  This library doesn't appear to exist in the ROS Hydro boost verions, 1.46.1, see [not here](http://www.boost.org/doc/libs/1_46_1/doc/html/atomic.html).

shaun-edwards, 2015-04-05:
@Jmeyer1292, please address ASAP, the build is broken under Hydro, see [here](http://jenkins.ros.org/job/devel-hydro-descartes/)

Jmeyer1292, 2015-04-06:
Looks like boost atomic became an official part of the library with version 1.53. Before hand it was an extension that one could manually install. There are a few possible solutions:

1. Include all of the boost atomic header files with Descartes
2. Remove the atomic operations and restrict our operation to single threads
3. Use compiler primitives. Both GCC and Clang support a __sync_fetch_and_add function that will do what we want. 

I'm leaning towards the latter solution, but it does mean relying on having specific compilers which is against the ROS specification (but so is c++11).

shaun-edwards, 2015-04-06:
What about using boost mutex to make it thread safe?

Jmeyer1292, 2015-04-06:
That works as well. I was excited to apply atomics, but I acknowledge that's not always the best justification for a design decision. In theory, atomics should be a few times faster but it probably won't matter to us. 

I assume we'll want to keep the branches (indigo and hydro) in sync? I will add a mutex to the generator class.

shaun-edwards, 2015-04-07:
Fixed in PR #101

closed 2015-04-07 by shaun-edwards