ros-industrial-consortium/descartes #103: TrajectoryPt cloneTo() method unimplemented

Jmeyer1292, 2015-04-06:
The base class for TrajectoryPt defines a virtual method cloneTo:

```c++
virtual void cloneTo(TrajectoryPt &clone) const
  {
    clone = *this;
    clone.setID(TrajectoryID::make_id());
  }
```

This method is unimplemented in all of the child classes and it will not do the 'expected' thing. What I believe it will do is 'slice' the 'this' variable and only copy the information that it is aware of in the base class itself. 

The fix is probably to create a clone function that returns an entirely new TrajectoryPtPtr with the appropriate underlying type. The implementation will probably look something like:
```c++
virtual TrajectoyPtPtr clone() const
{
// add cloning info for timing and ids
return new SomeTypeOfPt(*this);
}
```

Or it could by a cloneTo method. The difficulty there is memory management. You'd probably have to take a shared pointer by reference and call it's reset function.

shaun-edwards, 2015-04-07:
@jrgnicho...any ideas?

jrgnicho, 2015-04-07:
I'm on board with requiring each subclass to override the "clone()" method that returns a TrajectoryPtPtr that has been initialized with the an object of that particular class.

Jmeyer1292, 2015-04-09:
#105 addresses this issue

closed 2015-04-09 by Jmeyer1292