ros-industrial-consortium/descartes #114: AxialSymetricPt does not properly initialize tolerance frame in CartesianPt parent class

jrgnicho, 2015-04-16:
The AxialSymetricPt constructors assigns the nominal pose data to the wobj_base_member instead of the wobj_pt_ tolerance frame member.  Thus the correct translation and rotation bounds of the point aren't captured within the tolerance frame and they are set to zero instead.  This causes failures when calling the **getClosestJointPose()** method which in turn causes the SparsePlanner to fail at finding a path.

Jmeyer1292, 2015-04-16:
The issue is not with AxiallySymmetricPt. The issue (see #110) is that getClosestJointPose doesn't work with the 4 transforms  that can be specified in Cartesian point. Anyone who specifies more than just wob_pt_ will run into issues.

jrgnicho, 2015-04-16:
I agree that the getClosestJointPose() method is incomplete (see issue #43).  The proper solution would require using all transforms and evaluating the rotational bounds in quaternion space in order to avoid dealing with discontinuities inherent in the cartesian space euler angles.

closed 2015-04-16 by jrgnicho