ros-industrial-consortium/descartes #123: Installing descartes

bluefish89, 2015-05-11:
Hi!
Just want to ask if its possible to add some information about how to install the descartes package either in the wiki or the readme.
I would like to use it, but don't know how to install it.
Best regards

shaun-edwards, 2015-05-11:
@jrgnicho, looks like we forgot the most important part?  Can you add this info to the wiki....FYI, Descartes should be available in Hydro using `sudo apt-get install ros-hydro-descartes`.  Let me know if that doesn't work.

bluefish89, 2015-05-11:
For indigo apt-get install is not yet available I think. In my despair I also asked in ros-answers. Here is [link](http://answers.ros.org/question/208963/installing-package-from-source-in-indigo/) with the great answer from gvdhoorn.
Thanks for the help and best regards!

jrgnicho, 2015-05-11:
@shaun-edwards , I could not find ros-hydro-descartes as a debian package

gavanderhoorn, 2015-05-11:
We're all still waiting for Tully to do a sync of Hydro.

It's installable if you use `ros-shadow-fixed`.

130s, 2017-03-31:
I added [install section](http://wiki.ros.org/action/info/descartes/Tutorials/Getting%20Started%20with%20Descartes?action=diff&rev2=34&rev1=33) on the ros wiki. Please feel free to edit it as it's wiki, but this ticket may be closed.

closed 2017-04-06 by Jmeyer1292