ros-industrial-consortium/descartes #124: descartes with ikfast

filipmrc, 2015-05-12:
I've been struggling with getting descartes to work with IKfast. The getAllIK function always returns 1 solution which is equal to the start state of the robot. 

I think I have narrowed down the problem to the way  the IKfast moveit plugin works, when moveit::core::RobotState calls setFromIK, which then passes it on to ikfast_moveit_plugin::IKFastMoveitPlugin::SearchPositionIK which always picks the first solution if it is valid. Now I've checked and the solver always finds the same set of solutions , that made me think that always the same seed_state_ is generated, but that isn't the case. Could anyone help me out?

shaun-edwards, 2015-05-12:
@jrgnicho, can you respond.

jrgnicho, 2015-05-12:
@filipmrc , the behavior you described is consistent with the way that ikfast works.  The seed is ignored by ikfast so regardless of what you pass it will always return the first valid solution.  The ikfast implementation subclasses the kinematics::KinematicsBase interface in moveit, but unfortunately this interface doesn't expose all of the ik solutions even if it finds them (see [here](https://github.com/ros-planning/moveit_core/pull/231) for more on this issue).  For now, all I can suggest to use this [workaround](https://github.com/ros-industrial-consortium/godel/tree/hydro-devel/motoman_sia20d_descartes) for wrapping the ikfast into a descartes_core::RobotModel.  Alternatively, you could use the kdl solver which is a lot slower but it'll return multiple solutions when you call getAllIK().  
In regards to your first remark, are you referring to the **seed** as the start state of the robot?

filipmrc, 2015-05-12:
First of all thank you for your quick reply.

If I understood the code correctly, seeds are random joint states generated when descartes_core::RobotModel is initialized. They are then used as starting points for a solution by the solver. I am currently attempting the workaround you suggested so perhaps I will have more success with that.

jrgnicho, 2015-05-12:
Yes, the MoveitStateAdapter (a RobotModel implementation) behaves as you described.  However, it is meant to be used with the KDL kinematics plugin since the kdl numerical approach takes a seed in order to localize the search for a solution.

filipmrc, 2015-05-12:
There is just one thing that puzzles me. In my attempts to test descartes with ikfast I realized that all the returned solutions were equal to the current state of the robot simulation I started with moveit demo.launch . 

More specifically , when GetAllIK calls Get IK , in moveit_state_adapter.cpp:

if (robot_state_->setFromIK(robot_state_->getJointModelGroup(group_name_),tool_pose,tool_frame_))

After this line , robot_state_->copyJointGroupPositions returns the current joint values of the robot in the simulator. Before this line, it returns the random joint positions that it was set to. What could be the problem here? I suspect that getJointModelGroup(group_name_) somehow refreshes the joint positions to the current ones.

jrgnicho, 2015-05-12:
The robot_state_ member in the MoveitStateAdapter maintains its own internal state separate from the robot model in the running ros system.  Calling **setFromIK(...)** sets the internal state of the **robot_state_** member to a set of joint values that are valid for that cartesian pose of the tool; thus the subsequent call to **getJointModelGroup(...)** retrieves those joint values.  None of these operations are in any way synced to the "current robot model" managed by move_group.

filipmrc, 2015-05-17:
@jrgnicho I have a quick follow-up question regarding the workaround. Did you encounter any problems with pluginlib? More specifically, pluginlib loads the workaround class:
                        
    pluginlib::ClassLoader<descartes_core::RobotModel> r_model_loader("descartes_core", "descartes_core::RobotModel");
    descartes_core::RobotModelPtr ptr = r_model_loader.createInstance(PLUGIN_NAME);
    ptr->initialize(robot_description, group_name, world_frame, tcp_frame);

This returns the initialize function of the MoveitStateAdapter class. On the other hand , rtn->GetAllIK() returns the variant from the workaround class, but the class doesn't receive values from the call and instead gets junk. The code is the same as in the linked workaround, only names are changed.

jrgnicho, 2015-05-18:
That RobotModel implemented in that workaround is specific to that robot (the Motoman SIA20d) thus if you are using another robot then this won't work. 
Also, make sure that the "robot_description" ros parameter has been initialized with the urdf and srdf data from your robot's moveit config package;  the planning_context.launch file should do this for you.
Can you provide more details about what happens when you call **GetAllIK()**?

filipmrc, 2015-05-20:
I had put in all the right parameters for the code to work, at this time i have managed to make a version that doesn't use pluginlib and it works fine. Also in the original workaround, when I switched pluginlib with a standard "class factory" (I think that's what it's called, but I didn't know what a class was up until 2 weeks ago), the memory error when entering GetAllIK stopped, but MoveitStateAdapters initialize function was still called instead of the workarounds.

I diagnosed the GetAllIK error by printing the pose matrix in the main program and in the function when it was called. The values inside the function would regularly be junk (although the same numbers kept appearing, so that's a bit suspicious). As for the initialize problem, I read up on class inheritance chains and it all seemed to check out, so it might be the compiler.

jrgnicho, 2015-06-29:
@filipmrc did you ever get the pluginlib issue resolved? In case you weren't doing this, the instance should be **boost shared_ptr** to the Interface class when using the pluginlib mechanism.  See [here](http://wiki.ros.org/pluginlib#pluginlib.2BAC8-pluginlib_groovy.CA-f2705d99403652682a00e50252a7fd91f1196796_10) for an example of this.

BrettHemes, 2016-04-20:
Have there been any changes with respect to this issue since the last comments?  We are working on getting Descartes running with the Kuka Agilus series of robots and currently are experiencing the single-returned-solution issue when using IKFast.  Is the suggested [workaround](https://github.com/ros-industrial-consortium/godel/tree/hydro-devel/motoman_sia20d_descartes) still the best plan of action?

Jmeyer1292, 2016-04-20:
@BrettHemes With our current interfaces and Indigo, this is (unfortunately) the best plan of action.

gavanderhoorn, 2016-04-20:
Mea culpa (partly): I've not yet merged ros-planning/moveit_ikfast#57.

gavanderhoorn, 2016-04-20:
off-topic: @BrettHemes: may I ask where you got the Agilus urdf? I've recently added a support pkg for the R900 ([here](https://github.com/ros-industrial/kuka_experimental/tree/kr6_agilus_support)) based on a request I got on ROS Answers. I tried getting some feedback on whether it was all correct, but they never got back to me.

BrettHemes, 2016-04-20:
@gavanderhoorn Is it worth waiting for the merge of ros-planning/moveit_ikfast#57 to indigo-devel?

off-topic fyi: We are supporting one internally currently with some discussion underway to possibly make it public (i.e., via _kuka_experimental_ maybe?).  Currently we have _kuka_agilus_support_, _kuka_agilus_moveit_plugins_, and _\<kuka_variant>_moveit_config_'s for a subset of the Agilus series.  If there is interest I can start working on getting clearance to get this out (discussed briefly with @phvass recently).

BrettHemes, 2016-04-20:
@gavanderhoorn @Jmeyer1292 If I do proceed implementing the workaround, where would such package(s) live within the ROS/Kuka (meta)package hierarchy?

Jmeyer1292, 2016-04-20:
@BrettHemes We've always made them project-specific packages. There's no official place for these special Descartes models. At least not yet; It could be a good idea to start collecting them.

gavanderhoorn, 2016-04-20:
@BrettHemes wrote:
> @gavanderhoorn Is it worth waiting for the merge of ros-planning/moveit_ikfast#57 to indigo-devel?

I can't promise I'll be able to merge it this week. You can always take the PR and merge it locally. The rest of the infrastructure should be able to use the newer IKFast plugins afaik (@Jmeyer1292 ?), so it should work. Once the plugin is generated, there are no dependencies on `moveit_ikfast` anymore, so the version an 'end-user' has no influence on its correct functioning.

> off-topic fyi: We are supporting one internally currently with some discussion underway to possibly make it public (i.e., via _kuka_experimental_ maybe?).  Currently we have _kuka_agilus_support_, [...]

Right. if I understand correctly the Agilus is actually part of the KR6 series, correct? I've pushed a `kuka_kr6_support` pkg to `kuka_experimental`. If you could add your supported variants there that would be great. I'm definitely interested.

Jmeyer1292, 2016-04-20:
@BrettHemes @gavanderhoorn Even when/if the changes to MoveIt get merged, I will still have to do some things on my side to make the experience seamless.

gavanderhoorn, 2016-04-20:
Ok. That was the part I was missing.

Do you have any eta for those changes?

Jmeyer1292, 2016-04-20:
Well if you're serious about getting these changes in, then what I'm going to do is make a specialized version of our` MoveItStateAdapter` that overrides a couple of functions and calls the appropriate function in MoveIt's new `KinematicBase`.

I'll make it a priority, but all my Descartes work is free time, so I expect to have it proposed in about a week.

shaun-edwards, 2016-04-20:
@Jmeyer1292, we can get this done on project if you have the time.

Jmeyer1292, 2016-04-20:
@shaun-edwards I appreciate it. I'll work on it tonight/tomorrow and if things go smoothly, we can just use that time for review.

gavanderhoorn, 2016-04-20:
> Well if you're serious about getting these changes in, then what I'm going to do is make a specialized version of our` MoveItStateAdapter` that overrides a couple of functions and calls the appropriate function in MoveIt's new `KinematicBase`.

re: serious: I don't really have an option here: the rest of MoveIt already merged these changes into Indigo, so `moveit_ikfast` is the last missing piece.

Jmeyer1292, 2016-04-20:
@gavanderhoorn Fair enough. I apologize for my abruptness.

jrgnicho, 2016-04-20:
The changes in the [moveit_ikfast PR](https://github.com/ros-planning/moveit_ikfast/pull/57) are the same that were heavily reviewed before getting merged into jade-devel so there shouldn't be much reviewing needed.

BrettHemes, 2016-04-20:
Thanks for the support guys!  The Descartes plans we are getting so far look promising for our needs and getting this running faster will be critical for testing.

> I can't promise I'll be able to merge it this week.

@gavanderhoorn no problem. I will either merge the PR myself or live with KDL until then.

@Jmeyer1292 I look forward to the modifications you proposed!

Jmeyer1292, 2016-04-20:
@BrettHemes I'm glad there's promise. I'll get on the changes as quickly as I can. Let me know of issues as you run into them and we'll at least start working on them.

As a reminder to everyone (and future audiences), the IKFast plugin is as much about correctness as its about speed.

gavanderhoorn, 2016-04-21:
@Jmeyer1292 wrote:
> @gavanderhoorn Fair enough. I apologize for my abruptness.

No need to apologise.

gavanderhoorn, 2016-04-21:
@Levi-Armstrong wrote:
> As a reminder to everyone (and future audiences), the IKFast plugin is as much about correctness as its about speed.

I've always understood that IKFast has an average runtime of microseconds for queries, whereas KDL needs a couple of tens of milliseconds?

Jmeyer1292, 2016-04-21:
@gavanderhoorn While true, an additional benefit of IKFast is that it gives you all of the solutions for that point. We are not currently able to consistently explore every joint configuration for a given pose with KDL. This coverage of the entire solution space is important for getting consistent & minimal plans.

gavanderhoorn, 2016-04-21:
> This coverage of the entire solution space is important for getting consistent & minimal plans.

Certainly, but something like `trac_ik` is / will also capable of that. But afaik is quite a bit slower.

shaun-edwards, 2016-04-21:
Does `trac_ik` return *all* the solutions?  It's definitely better than KDL at being consistent, but it only returns the *closest* joint solution.

gavanderhoorn, 2016-04-21:
> Does `trac_ik` return *all* the solutions?  It's definitely better than KDL at being consistent, [..]

Doesn't currently do all solutions afaik. @pbeeson had an issue for that iirc.

> [..] but it only returns the *closest* joint solution.

That depends on how you configure it (see [trac_ik_kinematics_plugin/README.md](https://bitbucket.org/traclabs/trac_ik/src/a94c7e1ee1a30cf2b508f5e6a1f30b324af049bd/trac_ik_kinematics_plugin/README.md?at=master&fileviewer=file-view-default)).

pbeeson, 2016-04-21:
If you desire, it can return the first solution found, a single solution that best fits your IK conditions (minimum distance from seed, maximum manipulability) given the set of solutions found within the requested solve time, or that entire set.  In the upcoming release, if collision detection is enabled, only the first value in the set will be guaranteed to be collision free.

gavanderhoorn, 2016-04-21:
So 'all solutions' is now implemented?

pbeeson, 2016-04-21:
"Multiple solutions found within the allowed IK time."  That set may be large but is not exhaustive.  Like for a 7 DOF arm it may be hundreds of unique solutions, but there could be more.

The current solver formulations have no way to determine when the complete set of solutions has been found.

pbeeson, 2016-04-21:
But there is nothing special about how tracik does this.  Any seed based IK solver can be run multiple times with different seeds to get multiple solutions.

shaun-edwards, 2016-04-21:
I think this is the key difference.  For 6DOF arms, FastIK returns all the solutions.  This cannot be guaranteed with a numeric solver.  We try the multiple seeds approach with KDL, but I wonder how this will work better/differently with tracIK

pbeeson, 2016-04-21:
Shaun, in my experience with ikfast (using a Denso vs68) the solutions are
FAR from within the typical 1e-3 error we want. Take ikfast IK solution,
run through FK, and compare results to original pose. The results are
shockingly bad. Plus ikfast didn't fully obey joint limits. So personally I
don't trust it. For my application, I had to do a final run using
KDL/tracik with the ikfast solution as the seed (after clamping to joint
limits) to get good convergence within acceptable errors.

Jmeyer1292, 2016-04-21:
Well that's good to know.

shaun-edwards, 2016-04-21:
@pbeeson, do you think a hybrid fastIK/tracIK solution would work well?  Using fastIK to come up with seeds and tracIK to refine the solution?  What kind of performance hit would you predict versus fastIK alone?

pbeeson, 2016-04-21:
Minimal.  trac-ik's "Speed" mode, is sub millisecond.  

For refining this single solution where you start close to the goal, trac-ik's Non-Linear Optimization thread isn't going to "win out" over its "KDL" thread.   So, if you are going to do this, you can just as well use KDL until your error gets below the threshold you desire.   KDL's performance declines when the gradient descent search has one or more joints hit a limit (during the search 'simulation'), causing it to get stuck in local minima.  If you are running ikfast, you should be close to a "global" minima already so that KDL would just be refining that.   

Now, if you are using an ikfast solution that started outside/near/very-close-to joint limits, a final KDL pass may not converge if the minimum requires a joint to be outside its soft/hard/safety limits.  But trac-ik would, as it would search the entire space -- though your solution might be in some other portion of the space.

I hope all that made sense.

pbeeson, 2016-04-22:
Additionally @shaun-edwards, I'm not sure IKFast gives you ALL the answers.  For a 6-DOF robot, there could be MANY answers around the same configuration that give a Cartesian pose within acceptable errors.  Just tweak a joint by 0.00001 radians and you have a new answer.   Additionally, you can't let th eEnd Effector have "tolerances" like (don't care about roll, or orientation can be with 15 degrees on r,p,y), like you can in trac_ik (and also with some improvements to KDL).

shaun-edwards, 2016-04-23:
@pbeeson, that's fair.  Ignoring tolerances and imperfect math, I would expect a 6DOF robot to still have several solutions in all the different configurations.  For a traditional 6DOF industrial robot, IKFast returns 16 different solutions (this is pretty much what I mean by ALL).

pbeeson, 2016-04-23:
Ok.  Under that definitions, trac-IK probably would return "ALL" solutions too given 5ms or so runtime.  Furthermire they would be sorted by you condition to maximize/minimize.  If you tell me an exact URDF and ikfast IK build settings, I can try to run some quick comparisons of solve time versus number of solutions between the two sometime next week.  I could also let you know the error of the ikfast solutions.

Jmeyer1292, 2016-05-23:
The original issue should be addressed by the changes in #160 pending review of the `moveit_ikfast` PR for Indigo.

BrettHemes, 2017-02-03:
This can probably be closed now

closed 2017-04-06 by Jmeyer1292

yijiangh, 2018-02-18:
I think the trac-ik conversation here is not completely over yet. Under @shaun-edwards 's definition on _all ik solutions_:

> For a traditional 6DOF industrial robot, IKFast returns 16 different solutions (this is pretty much what I mean by ALL).

If `track-ik` is able to produce these solutions in 5 ms, according to @pbeeson ,

> Ok. Under that definitions, trac-IK probably would return "ALL" solutions too given 5ms or so runtime.

Then I think `trac-ik` deserves to be a good replacement for ikfast when ikfast configuration becomes troublesome for 7 dof robots (for example, 6-axis + linear track [here ](https://github.com/ros-industrial-consortium/descartes/issues/210)[1]). I'm interested in trying trac-ik with descartes but I have the following questions:

1. Does trac-ik supports `return all ik solutions` functionality now?

2. If it doesn't support `getAllIK` in its existing code base, how should I proceed? Can I just use trac-ik for moveit! kinematics solver, and simply use existing `moveit_state_adapter`? Or I need a subclass wrapper like we did for ikfast?

Based on @pbeeson 's description here: 

> But there is nothing special about how trac-ik does this. Any seed based IK solver can be run multiple times with different seeds to get multiple solutions.

I assume that we would use the `random seed start pose` strategy to trac-ik to get multiple ik solutions, which is already implemented in `moveit_state_adapter`.

Thanks!

[1]  https://github.com/ros-industrial-consortium/descartes/issues/210

pbeeson, 2018-02-19:
There is no guarantee that trac-ik will return **all** solutions, as there are infinite solutions for a 7-DOF robot and  potentially hundreds for some configurations of 6-DOF robots (consider a configuration where two inline rolls joints can move to anything so long as their sum is for example zero).    But I don't believe that any IK solver that would return **all** such solutions in finite time.

shaun-edwards, 2018-02-19:
Even for those arm kinematics, for which all solutions can be found, numerical solvers are not ideal.  @Jmeyer1292 and I tried this for standard industrial arms and were less than successful.  I believe there is a kinematics wrapper for KDL somewhere that tries to guess the proper seeds on startup.  Guessing the seeds for an arbitrary arm kinematics is not easy (but I'd love to be proven wrong).

jrgnicho, 2018-02-19:
@yijiangh I don't know that TrackIK would be a good replacement for IKFast as it does require a seed to produce different solutions as someone already pointed out.  It will most likely be faster than the default KDL numerical solver.  If you already have a TracIK plugin then it should be pretty straight forward to load it into the MoveitStateAdaptor as long as it is listed in the kinematics.yaml config file.