ros-industrial-consortium/descartes #130: Support of manipulators with more than 1 groups of joints

vkhamianok, 2015-07-09:
The "descartes" library fails in the function:

bool MoveitStateAdapter::isValid(const std::vector<double> &joint_pose) const

 with message:

/opt/ros/indigo/include/moveit/robot_state/robot_state.h:244: void moveit::core::RobotState::setVariableVelocities(const std::vector<double>&): Assertion `robot_model_->getVariableCount() <= velocity.size()' failed.

when I'm trying to calculate a Forward Kinematics using the function:

  virtual bool getFK(const std::vector<double> &joint_pose, Eigen::Affine3d &pose) const = 0;

for a manipulator with 2 (or more) groups of joints.

The problem is that active joints in a JointModelGroup named as "group_name_" is just a subset of all manipulators' joints. So number of active joints in JointModelGroup is less than VariableCount of a robot_state_ of such manipulator. So either this check fails:

  if (robot_state_->getJointModelGroup(group_name_)->getActiveJointModels().size() ==
      joint_pose.size())

if you pass a std::vector with positions for every joint_pose for a manipulator. Or:

    robot_state_->setVariableVelocities(std::vector<double>(joint_pose.size(), 0.));

fails (with the message shown in the beginning) if you pass a std::vector with positions just for a group. 

Seems that we need to change:
    robot_state_->setVariableVelocities(std::vector<double>(joint_pose.size(), 0.));
    robot_state_->setVariableAccelerations(std::vector<double>(joint_pose.size(), 0.));
to:
    robot_state_->setJointGroupVelocities(group_name_, std::vector<double>(joint_pose.size(), 0.));
    robot_state_->setJointGroupAccelerations(group_name_, std::vector<double>(joint_pose.size(), 0.));

But moveit::RobotState doesn't have such setJointGroupVelocities and setJointGroupAccelerations methods yet.

Jmeyer1292, 2015-07-09:
Yeah, this has been an issue that has come up a few times but I haven't gotten around to fixing it yet.

The temporary fix is to do the following:

Goto line 324 and 325 of moveit_state_adapter.cpp in descartes_moveit/

Remove the following portion of these lines:

joint_pose.size()

And replace it with the following:

robot_state_->getVariableCount()

So that line 324 should read:

robot_state_->setVariableVelocities(std::vector<double>(robot_state_->getVariableCount(), 0.));

vkhamianok, 2015-07-09:
The specified code:
```
robot_state_->setVariableVelocities(std::vector(robot_state_->getVariableCount(), 0.));
```
doesn't really help.
If I pass more positions values in the const std::vector<double> &joint_pose parameter than the group has. The first check:
```
  if (robot_state_->getJointModelGroup(group_name_)->getActiveJointModels().size() ==
      joint_pose.size())
```
fails with the message:
```
Size of joint pose: 10 doesn't match robot state variable size: 8
```

P.S. I've update tail of the initial message with another solution:
```
    robot_state_->setVariableVelocities(robot_state_->getJointModelGroup(group_name_)->getActiveJointModelNames(), std::vector<double>(joint_pose.size(), 0.));
    robot_state_->setVariableAccelerations(robot_state_->getJointModelGroup(group_name_)->getActiveJointModelNames(), std::vector<double>(joint_pose.size(), 0.));
    if (robot_state_->satisfiesBounds(robot_state_->getJointModelGroup(group_name_)))
```
Seems that it works as expected.

Jmeyer1292, 2015-07-10:
I apologize, I should have read your message more thoroughly before I responded off-hand like that. I think we need to do some re-imagining of the MoveIt state adapter to facilitate these more complex setups.

Can you share a little more about the robot you're using and what joint groups you've created so that we can effectively test on our end?

Jmeyer1292, 2017-04-06:
Nominally addressed by #164

closed 2017-04-06 by Jmeyer1292