ros-industrial-consortium/descartes #131: Errors in empty constructor

vkhamianok, 2015-07-09:
I have the following code:
```
descartes_core::RobotModelPtr robot_model_ptr;

MyDescartesPathPlanner::MyDescartesPathPlanner() :
		// Create a Descartes robot model
		robot_model_ptr(new descartes_moveit::MoveitStateAdapter()) {
	robot_model_ptr->initialize("robot_description", "manipulator", "/base_link", "tcp");
}
```
Here I invoke no-arguments constructor of the MoveitStateAdapter class and then initialize the instance with appropriate values. But because of missing of world_to_root_ initialization with Eigen::Affine3d::Identity() matrix in no-arguments constructor, the world_to_root_ has strange value. In my case matrix is:
```
[0] = {double} -3.1192113387569386e+90
[1] = {double} -nan(0xfd2b87fffffff)
[2] = {double} nan(0xfffffd2b87fff)
[3] = {double} 0
[4] = {double} -nan(0xfd2b87fffffff)
[5] = {double} nan(0xfffffd2b87fff)
[6] = {double} -3.1192113387569386e+90
[7] = {double} 0
[8] = {double} nan(0xfffffd2b87fff)
[9] = {double} -3.1192113387569386e+90
[10] = {double} -nan(0xfd2b87fffffff)
[11] = {double} 0
[12] = {double} -3.1192113387569386e+90
[13] = {double} -nan(0xfd2b87fffffff)
[14] = {double} nan(0xfffffd2b87fff)
[15] = {double} 1
```

Jmeyer1292, 2015-07-10:
+1 There's a lot of cleanup that needs to happen in this class. I will try and get this fixed and probably clean up the class to remove the non-default constructor (we should have one way to work the class correctly).

Thank you for taking the time to make these issues. I will get on this asap, but please feel free to submit PRs in the future if you'd like to. They're welcome.

Jmeyer1292, 2016-05-23:
#160 Addressed this issue. Thanks for bringing it ip.

closed 2016-05-23 by Jmeyer1292