ros-industrial-consortium/descartes #132: GitHub Builder

Manouchehri, 2015-07-17:
Since ROS already has Jenkins set up for Descartes, it shouldn't be too hard to add the [GitHub Pull Request Builder Plugin](https://github.com/janinko/ghprb). It'd be useful to make sure that our pull requests are passing.

I could also write a Travis CI script to build Descartes, which would be useful since anyone can opt to enable and build their own forks as well. The downside is that I'd basically be duplicating what Jenkins is already doing. Any input on this one?

shaun-edwards, 2015-07-17:
@Manouchehri, I've never used the Pull Request Builder plugin.  Can you point to other ROS projects that do?  I'm curious to see how it works.  In theory, I'm definitely on board with tools that help verify PRs.

I've seen other projects use Travis CI to build PRs (presumably similar to what you suggest above, but different).  A while back I tried to use Travis for building ROS packages.  I never succeeded, but I think it would be valuable.  If it's easy enough for you to create the script, then I'd be supportive.

Thanks so much for offering to contribute!

Manouchehri, 2015-07-17:
@shaun-edwards I don't know of any ROS projects/packages that do that, to be honest I haven't really been following any ROS projects until recently. I've had to deal with the nightmare of proprietary path planners before, so hearing about this project was pretty exciting. This along with ros_control seem extremely useful in keeping things vendor-neutral.

[Here's an example](https://github.com/siacs/Conversations/pull/1287) of Travis CI on a PR; it's handy to visually see on one page that you aren't going to break the entire build, and accept it with a click or two.

Jenkins isn't something I've had much experience with in the past, but I think that GitHub plugin is relatively straightforward. I believe the only concern is making sure you don't run a malicious PR by mistake (which is why the plugin requires a list of trusted users), although I don't think I've ever seen that happen on GitHub.

I doubt there's really any real technical advantages to using Travis CI over Jenkins (nothing wrong with running both either), but since ROS is an open source project, Travis CI is completely free. Anyone who forks a project can run a build, regardless if they're affiliated with ROS or not.

This weekend I'll play around a bit with compiling Descartes and I'll send over a PR when I get a Travis CI build script going.

shaun-edwards, 2015-12-23:
Closing this issue as Travis CI is now implemented across ROS-Industrial.

closed 2015-12-23 by shaun-edwards

Manouchehri, 2015-12-24:
@shaun-edwards Looks good! Might want to add a [Travis CI build status](https://docs.travis-ci.com/user/status-images/) icon to the `README.md`.