ros-industrial-consortium/descartes #141: Performance Benchmarks

Jmeyer1292, 2016-01-08:
In many cases, we are interested in measuring the "performance" of Descartes: That is to say, how "quickly" can we solve various problems that are representative of real-life situations?

To start down this road, I propose that we extend our unit tests to have a benchmark component. More specifically:
 1. Add a 6 DOF robot model with and without closed form IK
 2. Add a process path representative of a real situation, ideally with some documentation that describes the type of problem being solved.
 3. Measure the time of execution of this process many times and over many epochs. 
 4. Make a reasoned guess about the performance of the system. 
 5. [Hopefully] Record this performance and compare against previous benchmarks.

gavanderhoorn, 2016-01-10:
Would it perhaps make sense to keep this separate from any unit-tests?

I can see users wanting to run the unit-tests only once (or perhaps never, as it is supposed to be done by the developer(s)), but running benchmarks on different hw for instance, to gauge how much computing power is required to be able to use Descartes for a certain application.

Having to also set up and run all unit-tests seems undesirable in such a situation.

Jmeyer1292, 2016-01-11:
I think that's a great idea. Shaun suggested something like a `descartes_benchmarks` package that could house our models and code, and I think that would work well.

I still want some kind of performance-based regression test inside our main unit tests. I realize there are some complications related to build-configuration and the hardware you're running on, but I think we can work something out.