ros-industrial-consortium/descartes #143: core: not all licenses mentioned in package manifest

gavanderhoorn, 2016-02-18:
`descartes_core` only lists `Apache 2` as the license for the package ([here](https://github.com/ros-industrial-consortium/descartes/blob/6d3c38adfd1dcfc39dcb8044f636a2e1ecdeee22/descartes_core/package.xml#L11)), but at least [descartes_core/include/descartes_core/pretty_print.hpp](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_core/include/descartes_core/pretty_print.hpp) is actually `Boost Software License` licensed. REP-127 [requires](http://www.ros.org/reps/rep-0127.html#license-multiple-but-at-least-one) that packages list all licenses using multiple `<license>` tags.

The accompanying `LICENSE_1_0.txt` is also missing (but may not be required).

gavanderhoorn, 2016-02-18:
This may actually be an issue with other packages in `descartes` as well, but I haven't checked.

Jmeyer1292, 2018-07-11:
I deleted the pretty print file with the custom license, as it served very little purpose for all the code it had.