ros-industrial-consortium/descartes #146: ROS Dev build is failing

shaun-edwards, 2016-02-29:
The ROS dev build for Descartes is failing: http://build.ros.org/job/Idev__descartes__ubuntu_trusty_amd64/lastBuild/

gavanderhoorn, 2016-06-27:
A bit of a 'me too', but since I'm getting the same emails: it's still failing. Anything you need help with to fix this @Jmeyer1292?

Jmeyer1292, 2016-06-27:
Gijs,

Sorry about that. I figured I was only torturing Shaun with these e-mails. I've only briefly looked into this build error. I haven't ever been able to reproduce it anywhere but on the ROS build servers (I haven't installed docker, but I tried to follow all of the other build script steps from the error log). If you see what's going on, please let me know.

gavanderhoorn, 2016-06-27:
All contributors to the `descartes` packages receive those emails. At this point, that includes you, Jorge, Shaun, me, 'ros-industrial@domain.org' (:)), Isaac, Daniel, Ratnesh, P. Sturgeon and Brett.

> I haven't ever been able to reproduce it anywhere but on the ROS build servers

Looking at the error message, it could be a `build` or `test` depend that is missing when the tests are being compiled. Do you recognise the `moveit_state_adapter_test.cpp` it mentions?

Jmeyer1292, 2016-06-27:
Well that sucks. The file in question is [here](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_moveit/test/moveit_state_adapter_test.cpp). The "missing" header is [here](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/test/descartes_trajectory_test/robot_model_test.hpp).

The package with the header attempts to export it [here](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/CMakeLists.txt#L20). The package that consumes the header has a dependency on this package. 

I haven't been able to reproduce the error on my machines using catkin_make or catkin tools.

closed 2018-01-03 by Jmeyer1292