ros-industrial-consortium/descartes #147: Is it possible to lower the IK iterations for each way points?

yishinli, 2016-03-02:
```
[ INFO] [1456935580.602282360]: Found 4 joint solutions out of 10 iterations
[ INFO] [1456935580.603348158]: Found 4 joint solutions out of 10 iterations
[ INFO] [1456935580.604468261]: Found 4 joint solutions out of 10 iterations
[ INFO] [1456935580.605591487]: Found 3 joint solutions out of 10 iterations
[ INFO] [1456935580.606676848]: Found 4 joint solutions out of 10 iterations
[ INFO] [1456935580.606708753]: CartID: ID221 JointPoses count: 19

```
With DensePlanner, it takes 50 IK iterations for each way point.
Is it possible to lower the IK iterations for each way points?

Thanks and Regards.

Jmeyer1292, 2016-03-03:
@yishinli I apologize that I haven't been able to respond to your requestions these last couple of days. We're going through a consortium meeting at the moment. I'll get on the ball after tomorrow is done.

The "10 iterations" portion is unfortunately fixed at the moment. This is a "feature" of the KDL based kinematics that we use by default. You should not see it with an IKFast based solution that overrides the getAllIK() function.

The source code to change if you want to is at the top of [this] (https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_moveit/src/moveit_state_adapter.cpp) file.

yishinli, 2016-03-07:
@Jmeyer1292 - Thanks for your clarification. After lowering the SAMPLE_ITERATIONS, the iterations did decrease. Then, I noticed the following info: 
```
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: Found 1 joint solutions out of 1 iterations
[ INFO] [...]: CartID: ID6926 JointPoses count: 9
```
I traced into [cart_trajectory_pt.cpp](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/src/cart_trajectory_pt.cpp#L214), and learned it is related to the size of wobj_pt_ and tool_pt_. Two questions:

1. What is the acronym for [WCS](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/include/descartes_trajectory/cart_trajectory_pt.h#L321)?
2. Where do we setup the size for [tool_pt_](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/include/descartes_trajectory/cart_trajectory_pt.h#L320) and [wobj_pt_](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/include/descartes_trajectory/cart_trajectory_pt.h#L322)?

Thanks and Regards,

Jmeyer1292, 2016-03-10:
Hey @yishinli 

First the answers to your questions:
 1. I think WCS stands for world collision scene, but I'm not sure. At any rate, ```wobj_base_``` describes the fixed transformation from the origin of the world to the base of the tool.
 2. If by `size` you mean the number of poses which we generate for each point, then the appropriate argument to change will be the position and angle discretization variables and the tolerance range. See [here](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/include/descartes_trajectory/cart_trajectory_pt.h#L224).

What's happening is that we are attempting to numerically explore all of the "valid" positions for a given trajectory point. If you set discretization to zero, you should only have a single point as you've told Descartes to only solve for that exact point. If you set the discretization variables higher, then Descartes will sample in this space and attempt to solve IK for all of them.

I'd also have a look at [axially symmetric point] (https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/include/descartes_trajectory/axial_symmetric_pt.h) for an example of a point with open tolerances about a single axis. This point is implemented in terms of Cartesian Pt, so it's a good reference.

I'd also strongly advise that you consider using closed form IK, such as can be created by IKFast. See [this tutorial](http://wiki.ros.org/Industrial/Tutorials/Create_a_Fast_IK_Solution). If will greatly speed up your efforts. All serious work that we've done has used some form of closed-form kinematics and not the default implementation.

closed 2016-10-15 by yishinli