ros-industrial-consortium/descartes #151: Descartes Planning Spends a Lot of Time Calculating Edges

Jmeyer1292, 2016-03-18:
It appears that a large portion of the time spent building a planning graph is actually spent looking up joint positions from the joint solutions std::map. I believe we can refactor the `insertGraph` method to reduce this by passing the joint vectors generated during IK expansion directly to the edge builders.

shaun-edwards, 2016-03-19:
@Jmeyer1292, can you provide the call graph and the details of your test case here.  It would be ideal if you could provide the test code/procedure as well.

Jmeyer1292, 2016-03-21:
@shaun-edwards 

Here's a pretty typical example as produced by Valgrinds' callgrind tool. Note the right-most branch. Far more time is spent building edges than doing IK (at least when using some closed-form equivalent). The `linearWeight` function is doing the actual math but it is less than half of the cost of it's parent. Compiling under `RelWithDebInfo` indicates that this time is spent (mostly) searching the `std::map`.
[image]

Jmeyer1292, 2016-03-21:
As far as test code/procedure, I've built a little package [here](https://github.com/Jmeyer1292/descartes/tree/benchmarks). It essentially uses the Roscon-2015 demo with some tweaks. It needs some work before it's ready to put into our main-line branch.

Jmeyer1292, 2016-04-19:
#154 at least partially addresses this issue. I'm going to close this issue for the sake of clarity. Further performance questions will be continued in new threads.

closed 2016-04-19 by Jmeyer1292