ros-industrial-consortium/descartes #158: Post-Condition Checking

Jmeyer1292, 2016-04-12:
Currently, Descartes performs all of its validity checking at graph construction time. This means, for example, that collision checking is always performed on every single possible joint position. For many Descartes paths with somewhat fine discretization, this could mean hundreds of thousands of checks.

It _may_ be advantageous to perform these sorts of checks only after the graph has been built and solved. Each vertex of a candidate solution would be checked against some set of conditions and invalid ones would be tossed from the graph. If there were no corrections, the plan stands. If there were, then we re-run the plan. 

This could possibly save time in the case that most points aren't in collision. It also pushes the failure point back toward the plan retrieval stage.

shaun-edwards, 2016-04-13:
There seems to be pattern evolving in the Descartes planner that we may want to abstract out.

- Graph construction
  - Vertex generation/filtering
  - Edge generation/filtering
- Graph solving
  - Cost functions
- Solution processing
  - timing constraints
  - validity constraints (maybe just a generalization of timing)

A planner implementation could be an abstract process with different filtering/validity/cost methods applied at certain times.

BrettHemes, 2016-07-30:
@Jmeyer1292 +1 to implementing this.  I just sent an email to swri-pkg-dev but thought I would comment here as well.  I believe that my applications in particular would greatly benefit from this functionality.  I might have some time to work on this in the coming weeks so I will see what I can do.

@shaun-edwards Shaun, regarding your comment, are there any plans to implement such?  I feel like the resulting changes would be non-trivial and warrant some discussion beforehand.  Your thoughts?

Jmeyer1292, 2016-08-01:
@BrettHemes I have no idea how well this approach will actually work, but I think it's worth giving a shot. I'll try to mock up an implementation and I'll pose it here to discuss. If you have some time and it's important to you, I'd be really appreciative of help.

BrettHemes, 2016-08-02:
@Jmeyer1292 Yeah, I wouldn't put too much effort into this just yet.  I got a big performance increase with fairly little effort by multi-threading the joint solution computation and subsequent collision detection (see my comments in #55).  I think some more parallelization along with your #179 will be time better spent.

JeroenDM, 2017-05-09:
@Jmeyer1292 One disadvantage of performing validity checking after graph construction is that the graph becomes bigger and will take up more RAM. This is probably only relevant when there are a reasonable amount of collisions possible in the environment. But to me, the trade-off between calculation time and RAM usage seems important for graph based global optimization.

Jmeyer1292, 2017-05-09:
You're absolutely right. I've worked to reduce the cost per now and I think we can get it down some more. The recently merged changes drop memory usage by a large factor for sense graphs.

JeroenDM, 2017-05-09:
I didn't realize the most recent commits are added in the "godel_refactor" and "kinetic-devel" branches, sorry. Up until now, I was working with indigo-devel. I should probably upgrade to kinetic. I'm excited to check out the lowered memory usage :)