ros-industrial-consortium/descartes #180: CartTrajectoryPt::uniform() bug

davetcoleman, 2016-07-07:
I believe I've uncovered a bug in the discretization of poses in Descartes. I'm using AxialSymmetricPt and I find that I get basically the same discretization for both the X_AXIS and Z_AXIS options:

X_AXIS
[image]

Z_AXIS
[image]

Y_AXIS (for reference)
[image]

I've dug into the code for a long time and I believe the underlying method of using roll/pitch/yaw to discretize orientations at arbitrary increments and amounts is incorrect. I believe the 6-way for loop in the ``uniform()`` [function](https://github.com/ros-industrial-consortium/descartes/blob/indigo-devel/descartes_trajectory/src/cart_trajectory_pt.cpp#L118) is subject to errors such as Gimbal Lock... but I'm not exactly sure of the solution. I wanted to report this here as I'm calling it a night, and hopefully someone has some insight into this issue.

I believe the correct way to do this requires a big restructuring of this code - I think we need to call ``Eigen::AngleAxisd(rx, Eigen::Vector3d::UnitX()) `` in different ways depending on which axis we are discretizing.  Or maybe I just need to figure out how to use the tool discretization instead of just object discretization.

Have you guys run into something like this?

ClayFlannigan, 2016-07-07:
Yup, I think this is not the right way to sample orientations.  Uniform sampling of Euler angles produces 1) non-uniform distributions and 2) suffers from singularities (gimbal lock).  See [here](https://www-preview.ri.cmu.edu/pub_files/pub4/kuffner_james_2004_1/kuffner_james_2004_1.pdf).  Quaternions are the normal solution with [SLERP](https://en.wikipedia.org/wiki/Slerp) interpolation.

shaun-edwards, 2016-07-07:
There's been quite a few discussion threads here about sampling euler angles.  In additional to Clay's suggestions, #43 has some other references.  In the case of sampling about a single axis, the current approach should work (`AxialSymmetricPoint` enforces this).  My guess is that the constructor for `AxialSymmetricPoint` isn't working as expected.

davetcoleman, 2016-10-20:
I'm pretty sure I solved this problem in my ``bolt`` implementation of the Descartes planner - its essentially a much more moveit-centric way to create a graph of underconstrained redundant Cartesian poses. The function I wrote to turn tolerances on angles into candidate poses is [here](https://github.com/davetcoleman/bolt/blob/kinetic-devel/bolt_moveit/src/cart_path_planner.cpp#L234). It uses a similar [data structure](https://github.com/davetcoleman/bolt/blob/kinetic-devel/bolt_moveit/include/bolt_moveit/tolerances.h#L51) to what Descartes uses for tolerances, except I stored the values in a vector so that this [rotateOnAxis()](https://github.com/davetcoleman/bolt/blob/kinetic-devel/bolt_moveit/src/cart_path_planner.cpp#L306) recursive function can still work.

I'm wrapping up my Bolt work and need to desperately focus on my dissertation due in a month, so I can't integrate these fixes back into Descartes - especially since I stopped using Descartes a while back. But the references above should help whoever might be motivated to copy the functionality back into Descartes.