ros-industrial-consortium/descartes #190: How to avoid the robot collision in descartes

jasonvank, 2017-01-18:
I am using UR10 and Linux 14.04 Indigo. The purpose of my project is to perform masking. My robot also have the link of a box and a pico. I use the descartes tutorial to implement the [cartesian motion planning.](http://wiki.ros.org/descartes/Tutorials/Getting%20Started%20with%20Descartes)

When I used the motion planning through RViz, it can recognise the collision very well and all the joint is under the joint limit.
However, after I run the decartes, the planning will not recognise the collision anymore. Even the arm hits the box, it will continue to run.

Also, I have another question, regarding the pico, how can I change its direction and make it always in a downward direction. Since the pico always change its direction during the planning. 

Attached are the photos related my questions.
[direction of pico](https://s30.postimg.org/71ynx1tr5/IMG_0357.jpg)
[collision](https://s30.postimg.org/js2wa51pd/IMG_0354.jpg)

shaun-edwards, 2017-01-19:
Are your boxes in your URDF or are they added to the scene in some other way?

jasonvank, 2017-01-19:
Maybe I should rephrase Shaun.

I have included in my URDF the definition of the box and my tool (pico). I used the wizard setup assistant to include all my elements and define the group. Includes all the joints and my tool. I simply followed the tutorial.

This brings me to another question, I would like to learn how to define my TCP to be in the tip of my tool(pico)? That would solve a lot of problems for the planning I believe. 

As of now, I haven't been able to define the TCP in the tip of the tool, so I have to perform the Descartes motion having the TCP in the center of my last flange. 

Now coming back to the main issue, Descartes can't recognise my collision boundaries. Still plans my UR10 to go "across" my box. Yes, box is defined in URDF.

How would you recommend I could proceed from here?

Thanks from ARTC,
Jason.

shaun-edwards, 2017-01-19:
>This brings me to another question, I would like to learn how to define my TCP to be in the tip of my tool(pico)? That would solve a lot of problems for the planning I believe.

Add a frame (without geometry) to your URDF.  When defining the motion group in the wizard, use that new frame as the last link in the kinematic chain.

>Now coming back to the main issue, Descartes can't recognize my collision boundaries. Still plans my UR10 to go "across" my box. Yes, box is defined in URDF.

It's my understanding that Descartes can avoid collisions with items pre-defined in the URDF.  Have you told the robot model used in planning to check for collisions using [this](https://github.com/ros-industrial-consortium/descartes/blob/kinetic-devel/descartes_core/include/descartes_core/robot_model.h#L109) method?

shaun-edwards, 2017-01-26:
Is this still an issue?  Can this be closed?

closed 2017-01-27 by jasonvank