ros-industrial-consortium/descartes #191: Could not connect to action server

jasonvank, 2017-01-26:
I am using UR modern drivers to connect to my UR10 robot. 

When I run the descartes, it has this error as the picture below.

[error image](https://s30.postimg.org/p1lrqiy35/IMG_0430.jpg)

Could you provide some alternative solution please? Thanks!

gavanderhoorn, 2017-01-26:
@jasonvank: could you either:

 1. attach the image directly to the issue, or
 1. just copy-paste the error text from your console into your issue

There is no need to use a screenshot for something that is essentially text.

jasonvank, 2017-01-27:
@gavanderhoorn Thanks.

I have managed to solve the issue. 

When I run my descartes tutorial on my simulation, the robot moves as expected. However, I am trying to plan using the same solved points but the UR10 robot doesn't move (Yes, there's connection between the robot and ROS as I tried using a plugin planner before to test it). Is there any step I'm missing?  Maybe the joint values are not being published to the ToROSJointTrajectory function?

I'm using the Descartes tutorial.

Jason.

closed 2017-02-01 by jasonvank