ros-industrial-consortium/descartes #192: Set collision checking true

JeroenDM, 2017-02-14:
After finishing [the basic tutorial](http://wiki.ros.org/descartes/Tutorials/Getting%20Started%20with%20Descartes) I wanted to enable collision checking.
In the decartes_moveit docs I found the [isInCollision](http://docs.ros.org/indigo/api/descartes_moveit/html/classdescartes__moveit_1_1MoveitStateAdapter.html#a5d51aa83067d8ff101dccd5323d7b467) boolean, but it seems that the setCollisionCheck() method is not yet implemented as metioned on line 107 in the file moveit_state_adapter.h.

>   /**
>   * TODO: Checks for collisions at this joint pose. The setCollisionCheck(true) must have been
>   * called previously in order to enable collision checks, otherwise it will return false.
>   * @param joint_pose the joint values at which check for collisions will be made
>  */

In decartes_core, the RobotModel class has a [setCheckCollisions()](http://docs.ros.org/indigo/api/descartes_core/html/classdescartes__core_1_1RobotModel.html#a01e3298b9225961529a54087ee7ddf01) method (mention the naming difference with setCollisionCheck) When I try to add the following line to the tutorial1.cpp file
`model.setCheckCollisions(true);`
I get a compilation error when building (with catkin_make).
> ‘descartes_core::RobotModelPtr’ has no member named ‘setCheckCollisions’
   model.setCheckCollisions(true);

I tried adding this line before and after the initialize(...) is called in the tutorial code:
` model.setCheckCollisions(true);  // before`
`if (!model->initialize(robot_description, group_name, world_frame, tcp_frame))`
`{`
`  ROS_INFO("Could not initialize robot model");`
`  return -1;`
`}`
`// model.setCheckCollisions(true);  // or after`

I also don't understand where the Ptr comes from at the end of the descartes_core::RobotModelPtr used in the tutorial code. I'm using the indigo-devel branch, maybe the problem is related to differences between the branches..

Please forgive me for possibly making trivial mistakes, I'm relatively new to the ros universe.

JeroenDM, 2017-02-17:
I realize that I did make a trivial mistake, as predicted. I tried to call a c++ function using the dot operator. (I started learning ROS using python scripting...) I will use the double colon (::) operator, read more tutorials on virtual objects and pointers (that's where the Ptr comes from!) and then everything should work just fine.

Bart123456, 2017-02-22:
Instead of using the line
`model.setCheckCollisions(true);`
You can use the following line:
`model->setCheckCollisions(true);`

This has to do with the 'model' object being a pointer. I still have to look into the details myself, however.

closed 2017-05-17 by JeroenDM