ros-industrial-consortium/descartes #199: questions on ptr class reference

Thompson104, 2017-06-26:
I wonder if you guys could find class reference for the following?

descartes_core::TrajectoryPtPtr, specifically there is a "makeCartesianPoint" function
descartes_core::RobotModelPtr, specifically  there is a "reset" function

I found people are using these two functions but I'm not sure where I can find the class reference, I googled it for a long time. I wonder if you could help me with it?

Thank you

BrettHemes, 2017-06-27:
They are defined via a macro on line 28 of utils.h
https://github.com/ros-industrial-consortium/descartes/blob/44116b74045138a83009a0576457c7a7cc72e05b/descartes_core/include/descartes_core/utils.h#L28

The macro provides a quick way to create types from Boost shared pointers
from a provided class... this confused me as well at first.  This
particular macro is used for PathPlannerBase, RobotModel, and
TrajectoryPt.  Long story short, you can declare and use them as smart
pointers to objects of their respective types (
https://theboostcpplibraries.com/boost.smartpointers-shared-ownership).

Brett

[image: Inline image 1]

mitch1044444, 2017-06-27:
@BrettHemes Thank you for your help, this is great.

BrettHemes, 2017-07-03:
No problem @mitch1044444
Would you mind closing the issue if you are good on this?

Thompson104, 2017-07-03:
@BrettHemes you are too kind sir.

closed 2017-07-03 by Thompson104