ros-industrial-consortium/descartes #205: `catkin build` errors

mcevoyandy, 2017-10-23:
https://github.com/ros-industrial-consortium/descartes/blob/429a226b0d5bfab661df8cefaa01f1e0350f6473/descartes_moveit/package.xml#L28

Using `catkin build` I get the following errors:
```
WARNING: package "descartes_moveit" should not depend on metapackage "moveit_ros" but on its packages instead
WARNING: package "descartes_moveit" should not depend on metapackage "moveit_ros" but on its packages instead
[build] Package table is up to date.                                                                                  
Abandoned <<< descartes_moveit                                 [ Depends on unknown jobs: moveit_ros ]                
Abandoned <<< descartes_tutorials                              [ Depends on unknown jobs: moveit_ros ] 
```

Jmeyer1292, 2018-01-03:
I don't get the error and the automated build system seems to work. I'll make the suggested fix, but I wonder what the difference is.

VictorLamoine, 2018-04-19:
@Jmeyer1292 this issue is fixed, you can close it :)