ros-industrial-consortium/descartes #206: Serializing and deserializing ladder graph?

yijiangh, 2017-11-11:
Hi everyone,

I'm writing to ask for help and advice on the implementation of [serialization](https://en.wikipedia.org/wiki/Serialization) and deserialization of ladder graph.

In my [application](https://github.com/ros-industrial-consortium/descartes/issues/200), the ladder graph might take quite some time (up to 40 min) to compute. And this heavy computation makes debugging for any downstream process time consuming and hard.

Thus, I'm wondering if we can save (serializing) the ladder graph once it's computed and parse it back if needed? In this way, I can abort the program, change code, launch, deserialize the ladder graph and go directly to the bugs down the road.

I am thinking about creating a message type for ladder graph, and use similar techniques used in Godel's `[param_helpers](https://github.com/ros-industrial-consortium/godel/blob/7ce719d6f5537bda496f62b69b1caea0d4f0b3d7/godel_param_helpers/include/godel_param_helpers/godel_param_helpers.h#L8)` or `[trajectory library](https://github.com/ros-industrial-consortium/godel/blob/7ce719d6f5537bda496f62b69b1caea0d4f0b3d7/godel_surface_detection/src/services/trajectory_library.cpp#L6-L42)`. But since the ladder graph tends to be huge, this might not be a good solution.

Any comment or idea on this? If creating an msg is a feasible solution, any ideas on the design of the msg to make the serialization process more efficient? Could this feature be helpful for the broader user community of Descartes?

Thanks !

shaun-edwards, 2017-11-11:
I believe the underlying graph is a boost graph.  Is there any reason not to serialize the graph directly...assuming boost provides such functionality?

yijiangh, 2017-11-11:
@shaun-edwards Oh sorry, I forgot to mention that I'm working on `kinetic-devel`, where @Jmeyer1292 implemented a customized [ladder graph](https://github.com/ros-industrial-consortium/descartes/blob/429a226b0d5bfab661df8cefaa01f1e0350f6473/descartes_planner/include/descartes_planner/ladder_graph.h) without using the boost architecture...

yijiangh, 2017-11-12:
I wrote and tested a [descartes_msgs](https://github.com/yijiangh/descartes/tree/feature/serialize_ladder_graph/descartes_msgs) package and a [descartes_parser](https://github.com/yijiangh/descartes/tree/feature/serialize_ladder_graph/descartes_parser) package. They just wrap the ladder graph data into corresponding msgs to allow saving and parsing. The solution is not so elegant but I have tested them in my [framefab app](https://github.com/yijiangh/framefab_mpp/blob/2ca889ba753c629d0ed2250f12ea606b2cfc0e72/framefab_process_planning/src/generate_motion_plan.cpp#L303-L317) and they work pretty well.

Not sure if this issue and these prototyped solutions could be useful for anyone else, but any comment and suggestion is warmly welcomed!

closed 2017-11-12 by yijiangh