ros-industrial-consortium/descartes #214: Random offset in planned path

hansondon, 2018-03-08:
I just encountered a very weird problem when using Descartes with MoveIt!.
We have a fixed trajectory to plan and we always insert and start from the same pose. Every time the planned path can reproduce the trajectory perfectly but it may or may not add some offset to the path on a certain axis (X-axis in our case), completely at random. The screenshot below may give you a better idea:
[image]
[image]

The first photo is when the path follows our desired trajectory while the second one is a path that looks similar but about 10cm back on the X-axis. 

Our program is adapted from the tutorial. And we use the specific UR3 kinematics solver from ur_kinematics package. It is understandable that each time the planned path may be slightly different, but I really cannot figure out what could introduce such a great offset to the path randomly when all the conditions/inputs are the same? 

Any help would be appreciated!

hansondon, 2018-03-09:
Oops, it seems to be a bug in our getAllIK function.
Although I still have no idea why this bug can generate a different path every run, after some tiny fix it is working well now.
Sorry for the unnecessary issue report.

closed 2018-03-09 by hansondon

gavanderhoorn, 2018-03-09:
> Oops, it seems to be a bug in our getAllIK function.

@hansondon: this would be in a custom descartes wrapper for `ur_kinematics`?

hansondon, 2018-03-09:
@gavanderhoorn Yes. 
We checked [this issue](https://github.com/ros-industrial/universal_robot/issues/155) and it seems `ur_kinematics` is faster than IkFast solutions? Our code is adapted from the roscon2015 demo project so we are still using the `MoveitStateAdapter` to wrap it up.

BTW, is it actually better to set IkFast as the solver in the `kinematics.yaml` and use `IkFastMoveitStateAdapter` instead? (well, I know it may be inappropriate to ask another question in a closed issue, but this isn't an issue)

gavanderhoorn, 2018-03-09:
`ur_kinematics` provides analytic ik solvers for the various UR robots, written by hand by Kelsey. IKFast would be the same, but then generated. I haven't benchmarked the two, but I almost assume that `ur_kinematics` is comparable to IKFast, if not faster.

The only direct disadvantage I see of using `ur_kinematics` directly is that you'll have to maintain a wrapper, as you cannot reuse the `IkFastMoveitStateAdapter` with it.