ros-industrial-consortium/descartes #217: Problem with Changing Rotation Matrix along Path

demorise, 2018-04-11:
I'm using Descartes to plan a path in which the yaw component of the rotation matrix of points changes in the course of the motion. I noticed that with a constant yaw, descartes generates a valid path, but once the yaw value is varied along the path, descartes gives an error of the form: "No edges between user input points at index i and  i+1"  

e.g.
[ WARN] [1523422579.560259888]: No edges between user input points at index 425 and 426
[ERROR] [1523422579.561466246]: Could not solve for a valid path

With a little digging, I found out that the error occurs at points were there is a move between points with different yaw values (and hence different rotation matrices)

Please, any help on how to solve this will be highly appreciated. Thanks.

demorise, 2018-04-11:
In case anybody faces a similar problem:
So I found out the problem was from the "TimingConstraint(dt)" argument of CartTrajectoryPt. It was too small, so there wasn't enough time for the rotation to take place. I set it to -1, which caused it to be ignored, allowing the default joint velocity to be used for the rotation instead.

shaun-edwards, 2018-04-11:
@demorise, thanks for the explanation.  Am I right in assuming this can be closd?

demorise, 2018-04-12:
@shaun-edwards Yes. Any additional insight will however be appreciated.