ros-industrial-consortium/descartes #226: kinetic-devel branch build error due to "logError" function

JeroenDM, 2018-06-21:
I tried to build descartes on a clean installation of ROS kinetic and got the error:
```
/home/jeroendm/ros/ws_descartes/src/descartes/descartes_core/include/descartes_core/utils.h:83:63:
error: ‘logError’ was not declared in this scope
       logError("Invalid euler convention entry %i", convention);
                                                               ^
make[2]: *** [CMakeFiles/descartes_trajectory.dir/src/axial_symmetric_pt.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/descartes_trajectory.dir/src/trajectory.cpp.o] Error 1
make[1]: *** [CMakeFiles/descartes_trajectory.dir/all] Error 2
make: *** [all] Error 2
```

I noticed they have [changed `logError` to `CONSOLE_BRIDGE_logError`](https://github.com/ros/console_bridge/issues/18) in the [console_bridge](http://wiki.ros.org/console_bridge) package.

If this has to be fixed, I will do so and create a pull request.
Or should the function `ROS_ERROR` be used instead?

Or am I doing something wrong?

The complete error message can be found [here](https://gist.github.com/JeroenDM/d55f0ad444089fa625356a543ae485af).

ManMan88, 2018-06-24:
I also have the same issue (also for logInform).
For now I replaced logError with ROS_ERROR and logInform with ROS_INFO and then it compiles.

BrettHemes, 2018-06-27:
> I noticed they have changed logError to CONSOLE_BRIDGE_logError in the console_bridge package.

Ouch, this hit us as well.  I think the right way to proceed is to change all the calls to include new prefix.  +1 to a PR (unless @Jmeyer1292 thinks otherwise)

Jmeyer1292, 2018-06-27:
I'm cool with that. I've been on travel for a bit and I won't be back till tonight.

JeroenDM, 2018-06-28:
Apparently, I'm to slow, I just woke up this morning and the issue was fixed :) Thank you!

closed 2018-06-28 by JeroenDM