ros-industrial-consortium/descartes #234: Error when building descartes from source with melodic-migration branch

igorrecioh, 2019-01-10:
Hi all,
Environment: Ubuntu 18.04 with ROS Melodic.

I am following the instructions provided [here](http://wiki.ros.org/descartes/Tutorials/Getting%20Started%20with%20Descartes) but cloning the melodic-migration branch.

When I try
```
catkin_make_isolated
```

I obtain this errors:
```
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/melodic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_state/robot_state.h:41,
                 from /home/erle/ros_mara_ws/src/descartes/descartes_moveit/include/descartes_moveit/seed_search.h:38,
                 from /home/erle/ros_mara_ws/src/descartes/descartes_moveit/src/seed_search.cpp:1:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/erle/ros_mara_ws/src/descartes/descartes_moveit/src/seed_search.cpp:47:65:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/melodic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/melodic/include/moveit/kinematic_constraints/kinematic_constraint.h:40,
                 from /home/erle/ros_mara_ws/src/descartes/descartes_core/include/descartes_core/robot_model.h:23,
                 from /home/erle/ros_mara_ws/src/descartes/descartes_moveit/include/descartes_moveit/moveit_state_adapter.h:22,
                 from /home/erle/ros_mara_ws/src/descartes/descartes_moveit/src/moveit_state_adapter.cpp:21:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/erle/ros_mara_ws/src/descartes/descartes_moveit/src/moveit_state_adapter.cpp:157:67:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
CMakeFiles/descartes_moveit.dir/build.make:134: recipe for target 'CMakeFiles/descartes_moveit.dir/src/seed_search.cpp.o' failed
make[2]: *** [CMakeFiles/descartes_moveit.dir/src/seed_search.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/descartes_moveit.dir/build.make:86: recipe for target 'CMakeFiles/descartes_moveit.dir/src/moveit_state_adapter.cpp.o' failed
make[2]: *** [CMakeFiles/descartes_moveit.dir/src/moveit_state_adapter.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/descartes_moveit.dir/all' failed
make[1]: *** [CMakeFiles/descartes_moveit.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'descartes_moveit': 
  Command '['/home/erle/ros_mara_ws/devel_isolated/descartes_trajectory/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2
```
Seems that there is something wrong with
```
error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
```

I have tried searching something related but no success. Could anyone give help with this? Thank you in advance!

BrettHemes, 2019-01-11:
It is from this new MoveIt commit here https://github.com/ros-planning/moveit_tutorials/pull/255

This should be easily fixed by replacing all occurrences of `Affine3d` with `Isometry3d`

Running this in another repository of mine fixed the issue:
`find ./ -type f -exec sed -i 's/Affine3d/Isometry3d/g' {} \;`

jrgnicho, 2019-01-11:
@igorrecioh I think those instructions are outdated, `catkin_make` is no longer used as far as I know, try using catkin tools, so the build command would be `catkin build `

jrgnicho, 2019-01-11:
@igorrecioh are you building moveit from source?

igorrecioh, 2019-02-04:
I have seen that the changes needed for this have been merged! Thanks! This issue can be closed!

closed 2019-02-04 by igorrecioh