ros-industrial-consortium/descartes #243: tool_pt orientation_tolerance seems not working

jacknlliu, 2019-09-07:
We set the `orientation_tolerance` as 30 degree
```cpp
tool_pt.orientation_tolerance.z_lower = -30.0/180.0 * M_PI; // Search -30 degree to 30 degree
tool_pt.orientation_tolerance.z_upper = 30.0/180.0 * M_PI;
```
But the output trajectory, the z axis tolerance is larger than 30 degree, actually max 90 degree!

I set the cartesian trajectory point like
```cpp
    boost::shared_ptr<CartTrajectoryPt> pt(
          new CartTrajectoryPt(wobj_base, wobj_pt, tool_base, tool_pt, // Here we specify our frames
                               0, // Don't search in cartesian space. We want the points to be exact.
                              M_PI/90.0,
                               descartes_core::TimingConstraint(timing_constraint)));
```

jrgnicho, 2019-09-09:
The tolerance is applied relative to the tool point orientation, so it can still exceed that tolerance relative to a global reference frame's orientation

jacknlliu, 2019-09-09:
Is it relative to the `tool_base`, not the given `tool_pt` z-axis?

jacknlliu, 2019-09-10:
What about importing a new construction method of the `TolerancedFrame` like
```cpp
  TolerancedFrame(const Eigen::Affine3d &a, const std::vector<double> &pos_tol, const std::vector<double> &orient_tol)
    : Frame(a)
  {
    Eigen::Vector3d t = a.translation();
    Eigen::Matrix3d m = a.rotation();
    Eigen::Vector3d rxyz = m.eulerAngles(0, 1, 2);
    position_tolerance = ToleranceBase::createSymmetric<PositionTolerance>(t(0), t(1), t(2), pos_tol.at(0), pos_tol.at(1), pos_tol.at(2));
    orientation_tolerance = ToleranceBase::createSymmetric<OrientationTolerance>(rxyz(0), rxyz(1), rxyz(2), orient_tol.at(0), orient_tol.at(1), orient_tol.at(2));
  }
```

this will enable user create a tolerance frame with a `tolerance` about the given axis of the `tool_pt`

```cpp
std::vector<double> pos_tol(3, 0.0);
std::vector<double> orient_tol(3,0.0);
orient_tol.at(2) = M_PI/2.0;
TolerancedFrame tool_pt(point, pos_tol, orient_tol);
```

jrgnicho, 2019-09-10:
You can look into the sampling method for the cartesian point [here](https://github.com/ros-industrial-consortium/descartes/blob/065630c8711c71913503bc14069faca6aa9d30cc/descartes_trajectory/src/cart_trajectory_pt.cpp#L215) 

>> this will enable user create a tolerance frame with a tolerance relative to the given axis of the tool_pt

I believe that what you are requesting is already possible with the current implementation unless there's a bug