ros-industrial-consortium/descartes #248: Release for noetic

machinekoder, 2020-10-21:
The melodic-devel branch works nicely in noetic. Please release the package.

gy2256, 2021-02-07:
Cool, I am using noetic on Ubuntu 20.04 and will give it a try.

anilkir, 2022-04-22:
@machinekoder can you remember if you had to make any tweaks to build in noetic?

I am having compilation problems when trying to build on Ubuntu 20.04 with noetic, error is below:

CMake Error at /home/composites/catkin_ws/devel/share/descartes_core/cmake/descartes_coreConfig.cmake:113 (message):
  Project 'descartes_core' specifies '/include' as an include dir, which is
  not found.  It does neither exist as an absolute directory nor in
  '/home/composites/catkin_ws/src/descartes/descartes_core//include'.  Check
  the website 'https://github.com/industrial-moveit/descartes_core' for
  information and consider reporting the problem.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  descartes/descartes_trajectory/CMakeLists.txt:4 (find_package)

anilkir, 2022-06-23:
> @machinekoder can you remember if you had to make any tweaks to build in noetic?
>
> I am having compilation problems when trying to build on Ubuntu 20.04 with noetic, error is below:
>
> CMake Error at [...]

For anybody looking for a solution, it was a $PATH issue in my case. After adding to my path the below and rebuilding, issue was resolved. The solution is from https://github.com/ros/class_loader/issues/149#issuecomment-846873861

export PATH="/usr/bin/:/bin/:/sbin:$PATH"

thebirdgr, 2024-05-29:
This should be in the readme for anyone looking to test this for melodic.