ros-industrial-consortium/descartes #252: Issue of linking executable descartes_move

gy2256, 2021-03-26:
OS: Ubuntu 20.04
ROS: Noetic

I placed the descartes folder under src folder in my catkin_ws and ran `catkin_make`. I got the following error:

```
Base path: /home/xv/python_joy_ws
Source space: /home/xv/python_joy_ws/src
Build space: /home/xv/python_joy_ws/build
Devel space: /home/xv/python_joy_ws/devel
Install space: /home/xv/python_joy_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/xv/python_joy_ws/build"
####
####
#### Running command: "make -j12 -l12" in "/home/xv/python_joy_ws/build"
####
[  0%] Built target octomap_msgs_generate_messages_py
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  8%] Built target descartes_core
[  8%] Built target visualization_msgs_generate_messages_py
[  8%] Built target geometry_msgs_generate_messages_nodejs
[  8%] Built target geometry_msgs_generate_messages_cpp
[  8%] Built target visualization_msgs_generate_messages_nodejs
[  8%] Built target visualization_msgs_generate_messages_eus
[  8%] Built target visualization_msgs_generate_messages_lisp
[  8%] Built target moveit_msgs_generate_messages_py
[  8%] Built target object_recognition_msgs_generate_messages_lisp
[  8%] Built target actionlib_msgs_generate_messages_cpp
[  8%] Built target actionlib_msgs_generate_messages_py
[  8%] Built target std_msgs_generate_messages_py
[  8%] Built target std_msgs_generate_messages_nodejs
[  8%] Built target geometry_msgs_generate_messages_eus
[  8%] Built target tf2_msgs_generate_messages_nodejs
[  8%] Built target shape_msgs_generate_messages_cpp
[  8%] Built target octomap_msgs_generate_messages_lisp
[  8%] Built target shape_msgs_generate_messages_eus
[  8%] Built target std_msgs_generate_messages_lisp
[  8%] Built target roscpp_generate_messages_lisp
[  8%] Built target moveit_msgs_generate_messages_lisp
[  8%] Built target roscpp_generate_messages_cpp
[  8%] Built target geometry_msgs_generate_messages_py
[  8%] Built target tf2_msgs_generate_messages_eus
[  8%] Built target moveit_msgs_generate_messages_eus
[  8%] Built target sensor_msgs_generate_messages_py
[  8%] Built target geometry_msgs_generate_messages_lisp
[  8%] Built target trajectory_msgs_generate_messages_py
[  8%] Built target actionlib_generate_messages_eus
[  8%] Built target std_msgs_generate_messages_cpp
[  8%] Built target rosgraph_msgs_generate_messages_cpp
[  8%] Built target roscpp_generate_messages_nodejs
[  8%] Built target shape_msgs_generate_messages_nodejs
[  8%] Built target shape_msgs_generate_messages_py
[  8%] Built target rosgraph_msgs_generate_messages_py
[  8%] Built target moveit_msgs_generate_messages_cpp
[  8%] Built target moveit_msgs_generate_messages_nodejs
[  8%] Built target actionlib_msgs_generate_messages_lisp
[  8%] Built target std_msgs_generate_messages_eus
[  8%] Built target trajectory_msgs_generate_messages_cpp
[  8%] Built target _catkin_empty_exported_target
[  8%] Built target actionlib_msgs_generate_messages_nodejs
[  8%] Built target actionlib_generate_messages_py
[  8%] Built target actionlib_generate_messages_nodejs
[  8%] Built target visualization_msgs_generate_messages_cpp
[  8%] Built target sensor_msgs_generate_messages_cpp
[  8%] Built target sensor_msgs_generate_messages_eus
[  8%] Built target sensor_msgs_generate_messages_lisp
[  8%] Built target sensor_msgs_generate_messages_nodejs
[  8%] Built target rosgraph_msgs_generate_messages_nodejs
[  8%] Built target object_recognition_msgs_generate_messages_py
[  8%] Built target roscpp_generate_messages_py
[  8%] Built target trajectory_msgs_generate_messages_nodejs
[  8%] Built target object_recognition_msgs_generate_messages_cpp
[  8%] Built target octomap_msgs_generate_messages_eus
[  8%] Built target trajectory_msgs_generate_messages_eus
[  8%] Built target shape_msgs_generate_messages_lisp
[  8%] Built target trajectory_msgs_generate_messages_lisp
[  8%] Built target object_recognition_msgs_generate_messages_eus
[  8%] Built target object_recognition_msgs_generate_messages_nodejs
[  8%] Built target octomap_msgs_generate_messages_cpp
[  8%] Built target octomap_msgs_generate_messages_nodejs
[  8%] Built target dynamic_reconfigure_gencfg
[  8%] Built target dynamic_reconfigure_generate_messages_nodejs
[  8%] Built target tf_generate_messages_lisp
[  8%] Built target tf_generate_messages_nodejs
[  8%] Built target tf_generate_messages_cpp
[  8%] Built target dynamic_reconfigure_generate_messages_cpp
[  8%] Built target std_srvs_generate_messages_cpp
[  8%] Built target tf_generate_messages_eus
[  8%] Built target dynamic_reconfigure_generate_messages_eus
[  8%] Built target moveit_ros_manipulation_gencfg
[  8%] Built target tf_generate_messages_py
[  8%] Built target std_srvs_generate_messages_lisp
[  8%] Built target std_srvs_generate_messages_eus
[  8%] Built target std_srvs_generate_messages_nodejs
[  8%] Built target std_srvs_generate_messages_py
[  8%] Built target dynamic_reconfigure_generate_messages_py
[  8%] Built target moveit_ros_planning_gencfg
[  8%] Built target dynamic_reconfigure_generate_messages_lisp
[ 26%] Built target descartes_trajectory
[ 34%] Built target descartes_utilities
[ 60%] Built target descartes_planner
[ 82%] Built target descartes_moveit
[ 91%] Built target descartes_tests
[ 95%] Linking CXX executable /home/xv/python_joy_ws/devel/lib/descartes_move/descartes_move
/usr/bin/ld: CMakeFiles/descartes_move.dir/src/descartes_move.cpp.o: in function `main':
descartes_move.cpp:(.text+0x7a3): undefined reference to `makeTolerancedCartesianPoint(Eigen::Transform<double, 3, 2, 0> const&)'
/usr/bin/ld: descartes_move.cpp:(.text+0x8aa): undefined reference to `makeTolerancedCartesianPoint(Eigen::Transform<double, 3, 2, 0> const&)'
/usr/bin/ld: descartes_move.cpp:(.text+0xf15): undefined reference to `toROSJointTrajectory(std::vector<boost::shared_ptr<descartes_core::TrajectoryPt>, std::allocator<boost::shared_ptr<descartes_core::TrajectoryPt> > > const&, descartes_core::RobotModel const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, double)'
/usr/bin/ld: descartes_move.cpp:(.text+0xf24): undefined reference to `executeTrajectory(trajectory_msgs::JointTrajectory_<std::allocator<void> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [descartes/descartes_move/CMakeFiles/descartes_move.dir/build.make:188: /home/xv/python_joy_ws/devel/lib/descartes_move/descartes_move] Error 1
make[1]: *** [CMakeFiles/Makefile2:4449: descartes/descartes_move/CMakeFiles/descartes_move.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

JeroenDM, 2021-03-26:
Maybe you forgot to add it to the `CMakeLists.txt` of your custom package? You can find an example in the [descartes_tutorials](https://github.com/ros-industrial-consortium/descartes_tutorials/blob/master/descartes_tutorials/CMakeLists.txt). Some code snippets from this tutorial below.

Add it to the required packages
```cmake
find_package(catkin REQUIRED COMPONENTS
  roslib
  descartes_core
  descartes_moveit
  descartes_trajectory
  descartes_planner
  descartes_utilities
  # other packages
)
```

and link against it

```cmake
# Tutorial 1
add_executable(${PROJECT_NAME}_tutorial1 src/tutorial1.cpp)

target_link_libraries(${PROJECT_NAME}_tutorial1
 ${catkin_LIBRARIES}
)
```

gy2256, 2021-03-30:
Thanks a lot!

closed 2021-03-30 by gy2256